Snake-Edition/P32-FW #7: [ENHANCEMENT] Update to v4.5.0

espr14, 2023-01-11:


prongato, 2023-02-05:
I hope Prusa hurries and updates past 4.4.0 and 4.4.1, mesh bed leveling has a bug that is also there in the ported version.  I just tested ported Llama 4.4.1 and have already tested Prusa’s stock FW 4.4.0 - 4.4.1 and the mesh bed leveling is missing values after it finishes.  It’s easily viewable in the terminal of Octoprint or any other software that allows one to see the result of the mesh bed leveling grid output.  If anyone lists it as a bug on Llama, it’s not, it’s a big on the Prusa firmware.

espr14, 2023-06-06:
Versions 4.5, 4.6, and 5 currently do not work with MINI. Merging of MK4 and XL brought issues (at least) with GUI.

closed 2023-06-06 by espr14