Snake-Edition/P32-FW #8: [ENHANCEMENT] Show remaining time (relative) + end time (absolute) together

Eldenroot, 2023-01-11:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

Enter what type of printer or upgrade the enhancement applies to.
**Printer type** - [MINI]
**Optional upgrades** - Bondtech IFS, Wifi module

**Is your suggested improvement related to an existing problem? Please describe.**
  With internet connection you loose relative remaining time on print screen.

**Describe the expected functionality**
  It should display both values - something like Print ends: 08:15 (32 minutes left)...

espr14, 2023-01-11:
This is too long but it could be in two rows.

Eldenroot, 2023-01-11:
It was just an example... because my printer is connected to internet but with this I have lost the ability to use relative time... it sucks.

Maybe add a clock to the header?

espr14, 2023-01-12:
Header would be more complicated. But do you need both or a switch in a menu is enough?

espr14, 2023-09-10:
Fixed in 4.4.1.3

closed 2023-09-10 by espr14