Snake-Edition/P32-FW #9: [BUG] Skew Correct Isnt Working

Raider1284, 2023-01-13:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version** - 4.4.1

**Original or Custom firmware** - Llama 

**Describe the bug**
  Skew Correct isnt working, the dimensions of the test print stay the same

**How to reproduce**
  Print the calibration file, input the measurements and then reprint. The measured distances wont have changed

**Expected behavior**
  Everything should now be 60mm in distance but its the same as the original test print

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.

**Crash dump file**
  Please attach the crash dump file. This will make it easier for us to investigate the bug.

**Video**
  Please attach a video. It usually helps to solve the problem.

espr14, 2023-01-13:
Hotfix:
1. set the skew
2. restart the printer
3. print

I'll fix it so it will work without a restart.

Raider1284, 2023-01-13:
Thanks for the quick reply. I did try test prints both without restarting the printer and after restarting the printer. Unfortunately the dimensions aren't changing for the test print and it looks like skew control isnt working. I'm printing through octoprint if that makes any difference.

espr14, 2023-01-14:
I've tested it in a newer version where restart works. I'll fix the skew and release it.

espr14, 2023-01-16:
fixed in
https://github.com/Llama-FW/Llama-P32-FW/releases/tag/v4.4.02

closed 2023-06-06 by espr14