Snake-Edition/P32-FW #13: [BUG] fan doesn’t increase with changing percentage

prongato, 2023-02-05:
**Printer type** - [MINI]

**Printer firmware version** - 4.1.1

**Original or Custom firmware** - custom Llama

**Optional upgrades** - runout censor, Slice Mosquito hotend with Suunto 25mm fan

**USB drive or USB/Octoprint**  - Octoprint

**Describe the bug**
  When attempting to change the speed of the hotend fan, it doesn’t increase.  The fan will start/stop while cycling through the options but the speed never increases.

**How to reproduce**
  Attempting to change the amount of voltage supplying the hotend fan.

**Expected behavior**
  On the 4.3.4 version of Llama, the fan cycles like normal up to the 100% voltage and associated speed.

qp0n4ik, 2023-04-15:
I have the same problem for me, the RPM doesn't change when changing the fan setting while printing anything.  you could see it on the left side at the bottom. on 100% the fan still ~1600RPM 
 https://imgur.com/gallery/9YhCHlM

espr14, 2023-04-29:
Unfortunatelly, I don't have enough time to support this feature now. There tons of changes comming with WiFi, XL and MK4.

espr14, 2023-06-01:
I tried to fix this but max.fan speed stays still the same.

MiniUser2024, 2024-04-12:
Any update on this issue?  Increasing fan speed is a needed feature for me too.

espr14, 2024-04-12:
I still haven't merged everything from v4. Then I want to focus on HW and merge with v6 (will be much faster than in case of v5). Then I can try again the fan speed. The solution from v4 does not work so it needs different approach.

MiniUser2024, 2024-04-12:
Thanks for looking into this request.  Would looking at how Llama controls print fan speed be beneficial? Older firmware though.  https://github.com/matthewlloyd/Llama-Mini-Firmware

espr14, 2024-04-13:
That does not work. The approach could be still viable but requires deep analysis of the code changes between v4 and v5.