Snake-Edition/P32-FW #63: [FEATURE REQUEST] Change Filament Confirmation While Printing

jwidess, 2024-03-28:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

### Is your feature request related to an existing problem? Please describe.
Add a simple **confirmation** menu when clicking the "**Change Filament**" button during a print. I've accidentally selected this option **countless** time while changing Live Z or Speed parameters during a print. This issue has also been open since 2021 on the Official Prusa Buddy Firmware here, [ticket #1387](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1387)

### Describe the expected functionality
When clicking the "**Change Filament**" button during a print, a **confirmation** menu should appear to verify you want to pause the print and change filament, instead of immediately starting the change procedure.

espr14, 2024-04-02:
It was in 4.4 versions since [v4.4.1.4](https://github.com/Snake-Edition/P32-FW/releases/tag/v4.4.1.4). I'll get it back to 5.x versions.

jwidess, 2024-04-02:
> It was in 4.4 versions since [v4.4.1.4](https://github.com/Snake-Edition/P32-FW/releases/tag/v4.4.1.4). I'll get it back to 5.x versions.

Awesome! Thank you very much for your work on this firmware! I have been loving it.

jwidess, 2024-04-10:
@espr14 FYI might want to update the [README.md](https://github.com/Snake-Edition/P32-FW#:~:text=Change%20filament%3A%20Change%20of%20filament%20in%20Tune%20menu%20is%20moved%20to%20submenu%20to%20avoid%20unwanted%20interruption.) with this feature un-striked.  (Also #18, #7, and #9)

> 14. Change filament: Change of filament in Tune menu is moved to submenu to avoid unwanted interruption.