Snake-Edition/P32-FW #66: [BUG] Faster than intended speeds when changing filament

jwidess, 2024-04-30:
**Printer type** - Mini

**Printer firmware version** - Snake MINI FW 6.0.0.1

**Optional upgrades** 
- Filament Runout Sensor
- "[The Ultimate Prusa MINIKS+ Extruder](https://www.printables.com/model/624477-the-ultimate-prusa-miniks-extruder-input-shaper-co)" with extruder ESteps set to 142

**USB drive or USB/Octoprint**
 USB Drive

**Describe the bug**
  There seems to be some sort of bug with the "**Speed up (un)parking**" feature that results in changing filament movements sometimes happening **far faster** than intended. For example, when loading filament, the process of purging after loading the filament to the hotend can happen within a **second** as the speed is set extremely fast. This bug isn't consistent and stops after restarting the printer. Need to do more testing to figure out the cause, but I assume its something to do with the "Speed up (un)parking" feature.

**How to reproduce**
  Load filament after unloading, sometimes, the loading will happen at a very high speed which isn't intended. 

**Video**
  Will get a video sometime within the next few days.

jwidess, 2024-05-02:
Here's a video of the bug. Only seems to happen after unloading then loading filament. And after a restart purge and loading both operate at the correct speed.

https://github.com/Snake-Edition/P32-FW/assets/24278368/2b8d6efe-3bc6-42cd-a2a2-8bf22519761b

espr14, 2024-05-02:
Since the speed up is in XYZ only, this must be down in the original FW. Sometimes, I saw that some movements in loading/unloading didn't set the speeds and used the ones from the last print. After restart it uses the default values which are always safe and slow.

jwidess, 2024-05-25:
> Since the speed up is in XYZ only, this must be down in the original FW. Sometimes, I saw that some movements in loading/unloading didn't set the speeds and used the ones from the last print. After restart it uses the default values which are always safe and slow.

I'll have to do more testing to see if this happens in the stock firmware as well.

geekifier, 2024-08-16:
I am having the same issue as @jwidess. It did not happen in the original FW. Both the purge and the initial filament load are affected.

Czarnyszarf, 2024-10-23:
same error for me

Czarnyszarf, 2025-01-02:
I recompiled newest version on MINI and just added m852 gcode.. the issue still happens...strange,  it has something to do with these changes ?

espr14, 2025-01-07:
If you add M852 gcode support only, it should have no impact on E axis. I'm preparing version based on 6.2.0 alpha2 so we'll see if it still occurs. I'm not able to reproduce it. I use default and 570 steps/mm.

Czarnyszarf, 2025-01-26:
I installed the new firmware - still bug appears. I recorded everything, and pasted gcode for testing

The first filament load works right after turning on. Then started a test print with 3 filament changes. 

The unloading sequence is correct, loading - not so correct, it does a purge and slow load at correct speed but the travel is bugged, I suspect it has something to do with motor direction maybe? 

Please change .txt to .gcode
[universal wall mount hole_0.4n_0.2mm_PLA_MINIIS_8m.txt](https://github.com/user-attachments/files/18550143/universal.wall.mount.hole_0.4n_0.2mm_PLA_MINIIS_8m.txt)

Profile used: standard prusa 0.20 speed , here is my start gcode:
`M300 s800 P200 ;
M300 s800 P200 ;
M300 s800 P200 ;
M852 I-0.00374754 J0.0011723 K0.00248782 ;
M862.3 P "MINI" ; printer model check
M862.1 P[nozzle_diameter] ; nozzle diameter check
M862.5 P2 ; g-code level check
M862.6 P"Input shaper" ; FW feature check
G90 ; use absolute coordinates
M83 ; extruder relative mode
M104 S170 ; set extruder temp for bed leveling
M300 S440 P750 ;
M140 S[first_layer_bed_temperature] ; set bed temp
M109 R170 ; wait for bed leveling temp
M190 S[first_layer_bed_temperature] ; wait for bed temp
M569 S1 X Y ; set stealthchop for X Y
M204 T1250 ; set travel acceleration
G28 ; home all without mesh bed level
G29 ; mesh bed leveling 
M104 S[first_layer_temperature] ; set extruder temp
G92 E0

G1 X0 Y-2 Z3 F2400

M109 S[first_layer_temperature] ; wait for extruder temp
M300 S440 P3000 ; beep 3 sekundy
; intro line
G1 X10 Z0.2 F1000
G1 X70 E8 F900
G1 X140 E10 F700
G92 E0
M569 S0 X Y ; set spreadcycle for X Y
M204 T[machine_max_acceleration_travel] ; restore travel acceleration
M572 W0.06 ; set smooth time
M221 S95 ; set flow
`

Videos:
One video shows the experimental menu settings and firmware version along with first load after booting up. 
The second video shows the whole print along with the bug (~~4min in)

By the way - is it possible to slow down a homing after bed levelling and filament change by a bit? It seems a little too fast. 

https://github.com/user-attachments/assets/a1c8e725-56ea-41fa-90f9-cbcbaa4dcd16
https://github.com/user-attachments/assets/9e9d5369-8390-4b8f-8a36-0a5e55fd7282

espr14, 2025-01-26:
Movement around filament change is as fast as printer can. You can decrease max. speed of axes in slicer. Change it in the starting G code (use M203) or change the max. feedrate for X and Y in machine limits.

https://marlinfw.org/docs/gcode/M203.html

High speed is used to avoid filament oozing and model cooling.

Czarnyszarf, 2025-01-26:
okay, thanks - ill try the m203

Do you have any ideas why filament load is jamming ? Can you try recreating the issue on your printer using that gcode?

espr14, 2025-01-26:
Looks simple:

1. You have decreased E steps/mm from 325 to 142 which approx. halves the force on the filament (higher gear)
2. You haven't increased current of E motor to compensate it
3. MINI starts with acceleration of 4000 mm/s^2
4. In the G code you increase the acc. to 5000.

So you increase acceleration which requires more force while weakening the extruder.

espr14, 2025-01-26:
I use weak motor with 570 steps/mm while increasing current to 550 mA.

Czarnyszarf, 2025-01-26:
Then why it works fine on original firmware with same settings?  

I compiled my own firmware based on 6.1.2 using this commit as example wnich things to change: https://github.com/Snake-Edition/P32-FW/commit/be86dbbcc3ded91258b7afe7738ceea82656db1f

I modified Configuration_MINI.h , addmarlin.cmake , M852.cpp , planner.h and homing_utils.cpp -   it still appears. 

On unmodified firmware , it works fine

espr14, 2025-01-27:
If you have time, just try changing
`M201 X4000 Y4000 Z400 E5000 ; sets maximum accelerations, mm/sec^2`
to
`M201 X4000 Y4000 Z400 E4000 ; sets maximum accelerations, mm/sec^2`
in the provided g code (decrease acc. to 4000). I think this will make the filament change working. If it works then it's not a problem of the FW but the physical limits of the printer. Even if it works in original FW, the extruder is not capable of reaching defined acceleration and/or speed.

Czarnyszarf, 2025-01-27:
Tried it, didn’t work 

https://github.com/user-attachments/assets/0d2b5d16-13d1-442d-95dc-62c12df9981c

espr14, 2025-01-29:
I looked into configurations and it has the same jerk. So I see these options:

1. increase current
2. increase steps/mm (increase gearing)
3. decrease max. acceleration

espr14, 2025-02-06:
At the start of the printer the extruder is set to 4000 mm/sec^2 but the maximal acceleration is kept at 1250 only. In the PrusaSlicer profile it increases the extruder to 5000 and the maximal to 4000. Therefore the extruder acceleration is
- after printer start: 1250
- during and after printing: 4000
If you set extruder acceleration to 1250 in the PrusaSlicer profile, the filament change will work all the time.

Czarnyszarf, 2025-03-14:
Woudl it be possible to add simple "set acceleration to 1250" during filament change and revert it back to 5000 after it?

Czarnyszarf, 2025-06-13:
after your changes, the unloading works fine, only loading filament is broken.

What I discovered, the reason of broken filament change is..m852 command , if I run print without the M852 in start gcode- everything works fine, adding m852 breaks the script.

I don’t know about the config, maybe somehow it affects the E stepper? 

To made sure I tested two other extruders, the original mini and dual gear from mk3 (140 and 80 esteps) , both had same issue. 

The stepper makes a loud cracking noise and can be moved by hand ,