Snake-Edition/P32-FW #67: [ENHANCEMENT] mk3s+ support

OleUrgast, 2024-05-09:
The firmware works pretty well.
As Prusa has harmoniced all the actual firmwares for Mini, mk3.5, mk3.9, mk4 and xl, I wonder if it wouldn't be possible to use the mini's board and display to upgrade an old mk3s+ to sort of mk3.49.
Both z-motors could be connected in serial (like it is done on dual-z upgrade for the mini). For the heated bed a simple external mosfet should do.
If I compare both mk3.5 and mini, only crash detection and mmu3 support seem to miss at the moment.
As clone boards and display are available pretty cheap, this could be a quite benefitial cheap upgrade for the good old mk3s+ (esp. if it is already upgraded to Revo, orbiter extruder etc). Due to Inputshaping it would be quite a nice upgrade.
Just an idea...

espr14, 2024-05-10:
That could be easy to do. I'm just wondering what should be changed:
1. XYZ print size
2. MBL points
3. min. and max. temperatures
4. PID for hotend/bed

What else?

OleUrgast, 2024-05-10:
Filament loading sequence.
The mk3s+ has a direct drive, the mini has bowden. So the transport distance does not fit, extruder skipps if firmware trys to push the filament through the not existing bowden (have this problem also on my direct drive converted mini)
Also default input shaper parameters should fit the mk3.5

By the way: While the mmu3 would not work with the mini board, the chameleon3d should (as it is quite universal). So no real downside here.

So the only real downside using the mini board instead of the 3.5 upgrade at the moment would be the display orientation ;-) imho worth the difference in price.

However, as Prusa developed a tuneing routine for the XL to compensate more precisly for self induction of the motors to reduce vertical fine artifacts and this might be transfered to the mk4/mk3.9/mk3.5 in future, this might not be possible on the mini board due to different drivers. However, this might take years to be transfered to the mk3.5 anyways...

espr14, 2024-05-10:
5. filament (un)loading sequence
6. input shaper parameters

I would call it MK3.3 since 3.5 is way better upgrade.

espr14, 2024-06-20:
There were much more changes but it's done. Now you can build one ;-)
[https://github.com/Snake-Edition/P32-FW/releases/tag/v6.0.1.2](https://github.com/Snake-Edition/P32-FW/releases/tag/v6.0.1.2)

closed 2024-06-20 by espr14

espr14, 2024-07-03:
https://www.printables.com/model/931496-prusa-i3-mk33