Snake-Edition/P32-FW #69: [BUG] Faster nozzle cooling - Not working

jwidess, 2024-05-25:
**Printer type** - MINI

**Printer firmware version** - Snake MINI FW 6.0.1.1 - [Snake_MINI_6.0.1.1.bbf](https://github.com/Snake-Edition/P32-FW/releases/download/v6.0.1.1/Snake_MINI_6.0.1.1.bbf)

**Optional upgrades** - Filament Runout Sensor

**USB drive or USB/Octoprint**
USB Drive

### **Describe the bug**
When running the `M109 R170 C` command as I have setup to do in my "Start G-code" the nozzle waits to cool, however the part cooling fan **doesn't** turn on. From what I recall this was working pre firmware 6.0.

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.
```
M862.3 P "MINI" ; printer model check
M862.1 P[nozzle_diameter] ; nozzle diameter check
M862.5 P2 ; g-code level check
M862.6 P"Input shaper" ; FW feature check
M115 U5.1.0+13455
G90 ; use absolute coordinates
M83 ; extruder relative mode

M852 I0.001 J0.001 K0.003 ; SKEW CORRECTION

M104 S170 ; set extruder temp for bed leveling
M140 S[first_layer_bed_temperature] ; set bed temp

M109 R170 C ; wait for bed leveling temp - Added "C" for Snake FW "Faster nozzle cooling"

M190 S[first_layer_bed_temperature] ; wait for bed temp
M569 S1 X Y ; set stealthchop for X Y
M204 T1250 ; set travel acceleration
G28 ; home all without mesh bed level
G29 ; mesh bed leveling 
M104 S[first_layer_temperature] ; set extruder temp
G92 E0

G1 X0 Y-2 Z3 F2400

M109 S[first_layer_temperature] ; wait for extruder temp
...
```

espr14, 2024-05-26:
Update reverted it from 'C' back to 'F'. Will be fixed in next version and maybe in original Prusa FW as well.

espr14, 2024-07-11:
Fixed in 6.0.1.2

closed 2024-07-11 by espr14