Snake-Edition/P32-FW #71: [BUG] Mesh bed leveling is broken

xtomasnemec, 2024-07-15:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI]

**Printer firmware version** - [6.0.1.3]

**Original or Custom firmware** - [Custom]

**Optional upgrades** - Filament Runout Sensor (IFS)

**Octoprint**
  Please let us know if you print via USB flash drive or Octoprint

**Describe the bug**
  The first point of Mesh bed leveling is ina wrong place so the leveling fails. Stock firmware (6.0.3) doesn't do do this.

**How to reproduce**
 Go to Controll and run mesh bed leveling

espr14, 2024-07-15:
Works in my case. It's based on 6.0.1 but there should be no difference with 6.0.1.3 or 6.0.3. What the "wrong place" means? Too far on X, Y, or Z?

xtomasnemec, 2024-07-16:
> Works in my case. It's based on 6.0.1 but there should be no difference with 6.0.1.3 or 6.0.3. What the "wrong place" means? Too far on X, Y, or Z?

Too far on both -X and +Y

espr14, 2024-07-16:
In that case, I need to know the last Snake version where it worked and the first where it stopped working.

xtomasnemec, 2024-07-16:
It was working on 6.0.0.1 and it is broken on 6.0.1.3

espr14, 2024-07-17:
What about [6.0.1.1](https://github.com/Snake-Edition/P32-FW/releases/tag/v6.0.1.1)?

xtomasnemec, 2024-08-26:
> What about [6.0.1.1](https://github.com/Snake-Edition/P32-FW/releases/tag/v6.0.1.1)?

it is broken

xtomasnemec, 2024-08-28:
> Thanks for testing. Looks like it was caused by merging algorithm. I'll do clean update in the next version.

closed 2024-08-28 by xtomasnemec