Snake-Edition/P32-FW #72: [BUG] Y axis calibration failure

Hondaprogamer, 2024-08-27:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - [MINI longbed ]

**Printer firmware version** - [6.0.1 ...]

**Original or Custom firmware** - [custom]

**Optional upgrades** - [wife card, filament sensor, dual gear extruder , revo hot end ]

**USB drive or USB/Octoprint**
USB/octoprint

**Describe the bug**
  Found after installing WiFi card the y axis keeps failing the calibration test

**How to reproduce**
After firmware upgrade all calibrations worked fine and was printing normal. Now during y axis tests bed moves all the way back then 3/4 the way forward and fails 

**G-code**
  Please attach the G-code you had trouble with. This will make it easier for us to replicate the error.

**Crash dump file**
  Please send the crash dump file to Prusa by emailing it to reports@prusa3d.com. Sharing this file is important and helps us investigate the bug.

  Do not share the file publicly, as the crash dump contains a raw snapshot of the printer's memory and may include unencrypted sensitive information.

**Video**
  Please attach a video. It usually helps to solve the problem.

espr14, 2024-08-28:
Check the FW version in the Info -> Version Info. There should be `long bed` below the version. If not, you have the wrong version.

espr14, 2025-04-20:
Check Snake MINI FW 6.3.2.1. Just set proper Y axis length. If the homing stops sooner, check by hand if the axis goes smooth all the way to the end. If not, it's a HW issue.

espr14, 2025-05-17:
In FW 6.3.2.4 there's a fix for longer axes.