Snake-Edition/P32-FW #73: [BUG] mk3.3 wrong thermistor type in config?

OleUrgast, 2025-01-19:
### Printer model

mk3s+

### Firmware version

Snake Mini mk33 6.0.1.2

### Upgrades and modifications

zOrbiter 2.0 with Revo Six

### Printing from...

-not printing-

### Describe the bug

If I install the firmware for the mk3.3, the hotend themistor reports between 50 and 51°C while in reality it´s about 17°C.
It does not matter if I use the original thermistor or the Revo.
The moment I install the firmware for the "normal" mini, the reported temperature is correct.
If I heat up the hotend to about 230°C (reported on the display) it actually just reaches about 160°, so I can not load filament. However heating up to 280° I am just barely able to load PLA.
Maybe you set the wrong thermistor type in the config?

PS: Everything else seem to work nice. Good upgrade for the mk3 as soon as this issue is resolved.

espr14, 2025-01-19:
Thanks for report. I tried to set the proper thermistor but it seems it's the same as for the MINI so I'll fix it in the next release 6.2.0.1.

espr14, 2025-01-19:
How the bed sensor behaves? Is it correct?

OleUrgast, 2025-01-19:
Yes, Mini and mk3(s(+)) use the same thermistor. On both you can use a Revo heatercore as drop-in replacement (wich I realy like).
Heatbed thermistor also is the same.

For the release 6.2.0.1 please also correct coldend fan to 100% if possible. Prusa used Noctua fan at the original mk3, later replacing it by simular alternatives at 100% speed, but use a more powerful alternative fan on the mini at 38% speed. So without speeding up to 100% on the mk3 the fan is to slow and the firmware reports a coldend fan mailfunction.
I just discovered it yet, as I normaly connect the coldend fan gnd to permanent gnd anyway, so always 100% on. I did it also on all my minis. However, I only use rev 1.0 buddy boards, were the fan´s GND line is PWM controlled (so I circumvent the fan controll by conecting the fan to permanent GND). Newer Buddyboards controll the 5V-fanline by PWM instead, so to do the same you need a permanent 5V - wich the Buddyboard does not provide (and you need an step-down converter or soldering on the board).
So while the wrong fanspeed does not effect me, it might be a problem for other users.

OleUrgast, 2025-01-19:
Just detected the error in your config. You set "#define TEMP_SENSOR_0 55" Should be "#define TEMP_SENSOR_0 **5**" instead, same as for the mini.

espr14, 2025-01-19:
It's very hard to fix the fan. If you set 100%, it still runs at 40%. So you have to find the old (4.x) fan driver in FW, use the fix from Llama FW and solve conflicts with every new FW version.

OleUrgast, 2025-01-19:
OK. I just tell people to use a clone board from AliExpress (seems to be version 1.0 always) and show the "how to connect a quite fan to the mini" instruction I made some time ago or use a step-down converter to get a permanent 5V full power for the fan. 
Quite happily all the 7 boards I have are Rev. 1.0...

But if I remember correctly, the fan speed for the coldend fan isn´t set in percent. It is set to a value from 0 to 255. So if you set it to "100" you actually set it to 100/255, wich is about 40%...

OleUrgast, 2025-01-20:
Thanks, I installed the new version now and temperature for the hotend now is correct.

However I run into the next problems:
- The mk3, mk3.5 and the mk4 use 32 microsteps on the extruder. Wich should work fine with stock extruder (eStep/mm simply is doubled). However, for everyone using a BMG (about 620) or an Orbiter 2.0 (690)hotend, eSteps would become more than 1000 after doubling. Extended hardware menu alows a maximum of 1000 eSteps to be set. Using the minis buddy board 1/16 microstepping for x should be used, wich would work with the original extruder (halving eSteps) as this was the original setting of the mk3 (without s and +) anyway, but also with improved extruders.
- x-homing sometimes fails. Also x- and y-calibration always fails. This might be due to other TMCs used in Buddy than in xBuddy, so the values to detect collision to the frame should be the same as used for the mini (TMC2209), not the ones used for the xBuddy (TMC2130).
- ABL fails. Maybe the mesh is to big (5*5 or 7*7 on the mk3, only 4*4 on the mini) as the space on the Buddy is limited. But it might simply be an error because of unreliable homeing mentioned above, I am not sure yet.

You might add "(highly experimental)" to the filename for the update.

I will do some experiments next weekend and maybe will try some changes myself. I will report if I find a solution.

Many thanks for your great work.

OleUrgast, 2025-01-21:
All problems solved:

- eSteps over 1000 still can be set by start gcode. So with "M92 E1380" the extrusion is correct for the zOrbiter 2. However, as this isn´t stored permanently, I might recompile the firmware with 16 microsteps for E anyway. The reason for 32 microsteps on the mk3 was the "inconsistent extrusion issue", as 16 microsteps interpolated to 256 by the TMC2130 was one of the causes. So Prusa changed to 32 microsteps without interpolation. I am not even sure if it matters on TMC2209 anyway - and certainly no need to do so on geared extruders like BMG and orbiter. But maybe alternativly the limit in display settings can be increased to 2000?

- Homeing failures are due to wrong stall_sensivity, as I guessed. I use "M914 X145 Y145" in start g-code before homeing now - and homeing works without error. You might set these values in the config files for the mk33.
Actually the sensivity needed depends on the current set for the stepper. My first try was to reduce the current further - with 280mA on x sensorless homing worked also. However, x skipped while printing, so I returned to 300mA and played with the sensitivity instead. Results might vary a bit depending of the different printer (for example manufactorer of the bearings, lubrication etc.) but so far 145 seems to work.

- Bed leveling errors were result of inprecise homing - after homeing issue resolved, ABL works fine.

closed 2025-01-21 by OleUrgast

espr14, 2025-01-22:
Here's the FW for the MK3.3

https://github.com/Snake-Edition/P32-FW/releases/tag/v6.2.0.2