Snake-Edition/P32-FW #76: [BUG]

OggiCZ, 2025-03-17:
### Printer model

MK3S+ 

### Firmware version

6.2.0.

### Upgrades and modifications

bondtech + revo, filament sensor, bear frame ( moons motor )

### Printing from...

USB

### Describe the bug

Hello,

 first of all thank you for your work, but my mk3 wont work at all, any calibration fail, autohome fail on Y, everytime restart will happen. If i put bed towards, then home sometimes work, but if it is backward then never. If the connector is in good position then the bed goes forward, so probing point is in backward, is that ok? If i swap the connector bed goes backward as the original prusa mk3, but fail to home, its 6-8 time bounce and restart with fail on Y. Hotend fail to calibrate ( but i think that is because i got revo hotend ). Home XYZ fail only the Z is ok. My Z is 310mm so i put it in here, but when i move Z it goes only for 150 but it is on the end, not showing good height or bad Z steps? Homing may fails due moons motor, not original prusa or LDO. I swapped 3 boards and the issue is same. I cannot create my custom firmware because i dont know what to do. Sorry about my English, im older and i dont use any translator because i dont know if it translate good.

OleUrgast, 2025-03-17:
On my mk3.3 (with zOrbiter 2.0 and Revo) most worked well.
However, following points are important:
- The mk3.3 firmware uses 600mA for z-motors. This is correct, if you connect both z-motors in parallel. However, I connected them serial, so they get quite hot. A "M906 Z300" solved this.
- If using an Orbiter, you also need to set the e current accordingly: "M906 E600" was right for my motor.
- Buddyboard permanently sets coldend fan to 38%. Solution for a Rev1.0.0 board: Remove the black cable from the fan plug and put it directly to a permanent ground. Now the codend fan is permanently on.

If using different motors, the sensitivity needed to register a bump might differ. However, Marlin should allow to adjust it by using "M914" see https://marlinfw.org/docs/gcode/M914.html

Put the commands into a gcode file and "print" it. If you found the correct values, add the gcodes into your start gcode.

OggiCZ, 2025-03-17:
My problem is that I can't even get to the print, I can't homing. All axis calibrations fail. Only Z passes. The hotend is a revo, so that fails, but I expected that, what's worse is that I can't calibrate the axes or the first layer. It seems to me that the Y is swapped because if I have the original cable, the bed moves forward and the point where the sensor is is at the back. When I turned the cable, the bed moves backward and the sensor is as it should be on the mk3, but I don't know if it's correct. This needs to be solved directly in the FW. Unfortunately, visual studio only works for me on the original FW and this FW gives me python errors in the certificate and other errors, so I'm not able to compile it myself. I can easily send the exact name of the motor, the hotend is a revo 40W, at least I think it worked with the modified fw on the bondtech.

espr14, 2025-03-17:
To build the FW just run 'build.bat'. This will build all the variants and langugaes so remove what you don't need.

OleUrgast, 2025-03-17:
The axis calibration failing can be simply ignored. I do not know whats wrong there, but my mk3.3 printed fine. The "calibration" at the start just checks if the tension of the belts on x and y is correct. Would be nice if this would be fixed in firmware, but if you upgraded from a working mk3, you already know its correct, as you changed nothing at the mechanics. Maybe the firmware does not calculate the correct lenght between bumping left and right (or front and back) correctly by the dimensions set in firmware.

However, if homing does not work (so you can not do first layer calibration), skip this and do some gcode files with editor and just put the gcode into it with different sensitivitys, followed by homing gcode. Than try to print them, untill you find one were the sensitivity works. Continue, until you find the sensitivity wich does not work anymore. Then you use the value in the middle of the first and last value working.

As this value is valid until reset, you should now be able to do first layer calibration. You can set the date of the file into far past, so it always is shown at the end of the list if you sort files by date - and name it zzz.gcode, to see it last is sorted alphabeticly. So if you want to redo first layer calibration, just print this file before doing so.

Of course also put it into start gcode, you only need this file if recalibrating z-offset.

OleUrgast, 2025-03-17:
> To build the FW just run 'build.bat'. This will build all the variants and langugaes so remove what you don't need.

Actually I got also errors trying it.
Wich version of Python do you use and what other components are to be installed (wich c compiler)?
It´s a bit hard to find informations how to set up the enviroment...

espr14, 2025-03-17:
Homing sensitivity can be changed by M914. See this thread:
[https://github.com/Snake-Edition/P32-FW/issues/73](https://github.com/Snake-Edition/P32-FW/issues/73)

espr14, 2025-03-17:
I'm little bit lost. Try to change the sensitivity (M914) once you find proper values for X and Y let me know what else is not working.

OggiCZ, 2025-03-17:
Build.bat not working....i got many errors at pythoon 3,12 or 3,9....

  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.3568.0_x64__qbz5n2kfra8p0\lib\urllib\request.py", line 1389, in https_open
    return self.do_open(http.client.HTTPSConnection, req,
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.3568.0_x64__qbz5n2kfra8p0\lib\urllib\request.py", line 1349, in do_open
    raise URLError(err)
urllib.error.URLError: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: unable to get local issuer certificate (_ssl.c:1129)>

and many more. 

ESPR14: I cant home the printer. If i ran M914 before homing it will raise the same error. Y keeps bouncing to restart. X is OK.

OggiCZ, 2025-03-17:
Hurray, I figured it out, the cable at the Y was reversed as it should work on the mk3, it was a problem, connected like the others and it works, the first layer is ok, homing too, it overheats the hotend temperature a little more, but it will print, it's just that the filament sensor is not working properly, it detects it but nothing happens and it says to insert the filament into the filament sensor, I have a Bondtech, the sensor there is different than the original, so if it's the polarity, or why it detects it and then doesn't do anything, I don't know, I turned it off, I'm going to try printing via octoprint and we'll see, thank you very much everyone.

OggiCZ, 2025-03-17:
Which profile are you using on this modification? Mini with modified bed? Or 3,5? Please do someone have Prusaslicer profile for this build? And for everyone homing is different that mk3s so mistake is solved :)

OggiCZ, 2025-03-18:
Another problem....Z steps are doubled :( Z-Height original or my actually z-height doesnt corespond to reality :(

espr14, 2025-03-23:
Great. The closest profile should be i3 MK3.5. It should behave like MK3.5 because it has the same settings but it uses different drivers so the real performance could be different.

Original Z screws should be 4x 2 mm pitch. FW is set to 400 microsteps per millimeter with 16 microsteps. That's 3200 (micro)steps per revolution. 1 revolution is 8 mm which gives 400 steps/mm. If your Z axis is 2x as fast then you should add `M92 Z800` or 'M92 Z200' (if slow).