Snake-Edition/P32-FW #77: [BUG] Pid autotune (m303) doesn't work

Czarnyszarf, 2025-04-16:
### Printer model

Mini

### Firmware version

6.2.0.4

### Upgrades and modifications

stock

### Printing from...

octoprint

### Describe the bug

When trying to do a PID tune printer responds with "Unknown command: "M303" "

For example, I tried M303 E0 C8 S240

### How to reproduce

Type M303 E0 C8 S240

### Expected behavior

printer doing a pid tune..

espr14, 2025-04-16:
PID was removed from the original FW. It won't be trivial to put it back.

espr14, 2025-04-28:
I've put some controls for PID in the new version, however, I haven't tested it yet.

[https://github.com/Snake-Edition/P32-FW/releases/tag/v6.3.2.3](https://github.com/Snake-Edition/P32-FW/releases/tag/v6.3.2.3)