Snake-Edition/P32-FW #78: [FEATURE REQUEST]

Lino77, 2025-04-21:
### Printer model

MK3.5

### Describe the feature

Hello I am currently building a CoreXY based on Mk3.5 is it possible to adapt your firmware to it

espr14, 2025-04-21:
No problem. I will release a starting version based on MK3.5. But then I need to provide proper homing sensitivities and motor currents. Since coreXY has more friction, the sensitivity has to be higher and the current as well to handle high accelerations at high speeds.

Lino77, 2025-04-21:
That's great. I'm planning on building a Core One that looks like a Prusa XL, only without the Nextruder and toolchanger, but with a Revo PZ-Probe. I already have everything except the extension board for the Core One, the 0.9 stepper for XY. So there will be a mix of Core One, XL, and either the Stealth or EVA3 as the toolhead. I'm not sure yet.

espr14, 2025-04-21:
I've created XL size printer from Prusa MINI.

Here's the i3 MK3.5 coreXY FW: [Snake MINI FW 6.3.2.2](https://github.com/Snake-Edition/P32-FW/releases/tag/v6.3.2.2)

Lino77, 2025-04-22:
Oaky firmware can be flashed and steppers can rotate. I can report more at the weekend.

espr14, 2025-04-28:
How's it going? I've released new version with more adjustments that could be handy.

[https://github.com/Snake-Edition/P32-FW/releases/tag/v6.3.2.3](https://github.com/Snake-Edition/P32-FW/releases/tag/v6.3.2.3)

Lino77, 2025-04-28:
it's alive, I'm still setting it up, had various problems with homing

https://youtube.com/shorts/pWlh-Li6htw?feature=share

Lino77, 2025-04-29:
I'm having trouble with the fan self-test. Could it be that version 6.3.2 isn't quite ready for the MK4 and MK3.5 yet? The test fails and runs differently than on my MK3.5 and MK4s.

espr14, 2025-04-29:
Version 6.3.2 was released for Core ONE only so there may be things that does not work properly for other printers. In case of fan selftest you should try 6.2.4.

Lino77, 2025-04-29:
adaptive mesh doesn't work either

espr14, 2025-04-30:
Let's wait for 6.3 update for other printers. However, I haven't seen any in print issue on MINI.